uugear/Witty-Pi-4 #4: No internet sync in daemon.sh

tennisparty, 2023-02-24:
On WittyPi3 there was a call in the daemon.sh file to a sync time script - [https://github.com/uugear/Witty-Pi-3/blob/49763e0487e941ecbc06430368355002f3d3463b/Software/wittypi/daemon.sh#L108](url) This sync time script had a call to sync the pi with network time if there was internet - [https://github.com/uugear/Witty-Pi-3/blob/49763e0487e941ecbc06430368355002f3d3463b/Software/wittypi/syncTime.sh#L40](url)

In the WittyPi4 daemon there is no call to sync the time with the internet. This means a user has to manually sync the time when the board has been re-attached.

uugear, 2023-02-24:
This change is made on purpose. The accuracy of RTC is good enough and there is no need to sync the time during every boot. On the other hand, we have seen many cases that the time was incorrectly synchronised due to a network issue, and hence causes problem on subsequent schedule script execution.

tennisparty, 2023-02-24:
Ok thanks that is useful to know. I'll triple check whether that works for me because my board was 4 hrs out when plugged in this morning. I'll try and get to the bottom of that

closed 2023-05-25 by uugear