uugear/Witty-Pi-4 #5: reading of charge percentage

dmasar, 2023-03-11:
Hi, i would like just to ask wheter it is possible to read percentage status of battery charging or charge level. thanks.

uugear, 2023-03-14:
This is unfortunately not possible. During the charging procedure, the battery is actually disconnected from the load (DC/DC converter) and the input voltage measured by the MCU is actually from the type-C connector. Without knowing the actual voltage of the battery, there is no way to estimate the charge percentage.

closed 2023-03-14 by uugear

matthiasg, 2023-11-14:
what about when it is not charging and in the process of discharging ?

uugear, 2023-11-15:
> what about when it is not charging and in the process of discharging ?

If the power supply on USB-C is disconnected, Witty Pi 4 L3V7 can know the current battery voltage (and maybe estimate the remaining capacity).