uugear/Witty-Pi-4 #8: Seems Witty Pi board is not connected?

asztalosdani, 2023-05-20:
Hi,

I have a **Witty Pi 4 L3V7** connected to a Raspberry pi zero W, the power is connected to the Witty, I can turn the raspberry on/off using the button.
I installed the witty software following the documentation, but running `./wittyPi.sh` results in 
```
================================================================================
|                                                                              |
|   Witty Pi - Realtime Clock + Power Management for Raspberry Pi              |
|                                                                              |
|            < Version 4.13 >     by Dun Cat B.V. (UUGear)                     |
|                                                                              |
================================================================================

Seems Witty Pi board is not connected? Quitting...
```
And indeed running `i2cdetect -y 1` outputs
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```
I confirmed i2c and serial port is enabled in `raspi-config`. What else should I try?

uugear, 2023-05-22:
Most probably it is due to unreliable connection. This is more likely the case when using it with Raspberry Pi Zero W -- please double check the header soldered on Raspberry Pi Zero W (especially for SDA and SCL pins).  It has to be well soldered, and just putting the header in place will not be enough.

uugear, 2023-05-23:
Older version of software might wrongly detect the board and hence try to communicate with it and prints out the Read failed errors. To you the results are the same with both SD cards: the board is not detected.

However it doesn't seem to be the firmware issue too. As you can turn on your Pi by tapping the button on Witty Pi 4, it means its firmware is functioning well. It is the same firmware that emulate an I2C device with address 0x08, which should also be detected.

Can not rule out the possibility that your Raspberry Pi's I2C bus is faulty (we have saw several cases like that). It is recommended to try it with another Raspberry Pi to narrow down the issue.

uugear, 2023-05-25:
You may give it a try but the information we have do not point to that direction: the firmware seems working already.

asztalosdani, 2023-05-25:
What else should I try? Is it a defective board?

tomjn, 2023-09-16:
Noting I'm in this exact situation with a Witty Pi 4 mini + Pi Zero 1 W. I've used a hammerhead GPIO pin header from pimoroni + extender pins to connect it and receive the same output from the debug commands. The board still powers the Pi though.

Something to note though, while I'm going to experiment with other headers to try and improve the pin connection, it might be nice if the software said this so I didn't have to google and read through github, especially since the message is a rhetorical question

This:

```
Seems Witty Pi board is not connected? Quitting...
```

Would be better with something like this:

```
We cannot detect a connected Witty Pi board, please attach a board and double check the header connection is secure if one is already attached. Quitting...
```

Bonus points if it can directly link to a debugging/troubleshooting doc

tomjn, 2023-09-17:
PR created to do just that: https://github.com/uugear/Witty-Pi-4/pull/13

I couldn't find a suitable doc to link to, but I did fix some tab/space mix issues and mentioned the version of the board the script is intended for in the output since I noticed a lot of the tech support threads ask this and it'd be unnecessary if the script just stated it

closed 2024-05-13 by uugear