uugear/Witty-Pi-4 #9: Unreliable I2C access on Raspberry Pi 2

martinling, 2023-05-25:
I have a Witty Pi 4 attached to a Raspberry Pi 2, running a freshly installed Raspbian. I installed the software with `https://www.uugear.com/repo/WittyPi4/install.sh` and rebooted the Pi.

Currently the Pi 2 is being powered from its own 5V input. There is a CR2032 cell installed in the Witty. Nothing else is attached to the GPIO header or to the Witty Pi.

It seems that accessing the I2C interface does not work reliably:

```
$ ./wittyPi.sh 
================================================================================
|                                                                              |
|   Witty Pi - Realtime Clock + Power Management for Raspberry Pi              |
|                                                                              |
|            < Version 4.13 >     by Dun Cat B.V. (UUGear)                     |
|                                                                              |
================================================================================
Error: Read failed
Error: Read failed
Error: Read failed
>>> Current temperature: 24°C / 75.2°F
>>> Your system time is: 2023-05-25 13:08:02 BST
Error: Read failed
Error: Read failed
Error: Read failed
Error: Read failed
```

Running `i2cdetect` reliably shows the device at address 0x08:
```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

However, reading a single register with `i2cget` only works about 25% of the time:
```
$ i2cget -y 1 8 0
0x26
$ i2cget -y 1 8 0
Error: Read failed
```

Using `i2cdump` will read out most of the registers successfully, but 4-5 consecutive reads will always fail, as indicated by the `XX XX XX XX XX` entries here:

```
$ i2cdump -y 1 8
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 26 05 0a 05 01 01 48 00 00 00 00 03 01 00 00 00    &?????H....??...
10: 08 00 04 ff 64 32 ff 00 14 14 00 00 00 00 00 00    ?.?.d2..??......
20: 00 00 00 00 00 76 01 00 00 00 00 00 00 00 00 00    .....v?.........
30: 00 00 17 00 4b 50 00 00 76 00 d8 09 13 25 04 05    ..?.KP..v.???%??
40: 23 80 80 80 80 80 00 18 00 00 00 00 00 00 00 00    #?????.?........
50: 00 00 00 00 00 00 00 00 01 00 01 00 01 00 00 00    ........?.?.?...
60: 01 11 00 30 02 4f 0b 99 07 00 00 04 01 00 00 e8    ??.0?O???..??..?
70: 03 00 00 00 00 00 00 75 01 14 01 00 00 e8 03 00    ?......u???..??.
80: 00 00 00 d2 00 00 00 00 00 a7 00 00 03 01 00 00    ...?.....?..??..
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
a0: 00 00 00 00 00 XX XX XX XX XX 00 00 00 00 00 00    .....XXXXX......
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 70 01 40 01 25 02 0b 02 8d    .......p?@?%????
d0: 01 8c 01 00 00 00 00 bb 06 66 02 60 02 4f 02 3e    ???....??f?`?O?>
e0: 02 8c 01 00 01 e5 00 57 00 00 01 48 87 0a 5d 00    ???.??.W..?H??].
f0: 00 00 01 78 00 00 19 35 be 02 08 e1 56 e9 03 01    ..?x..?5????V???
```

The issue is not with specific registers, these `XX` groups may occur anywhere in the register map.

Any ideas?

uugear, 2023-05-25:
If your Pi 2 was powered by its own power supply, it boot directly and you most probably did not tap the button on Witty Pi 4 (so its white LED blinks), is it the case?

When the white LED blinks, Witty Pi 4's MCU is in sleep mode, and gets woke up every few seconds (for blinking the LED etc.). If your I2C request comes when MCU is awake, it gets processed properly, otherwise not.

martinling, 2023-05-25:
Ah OK, that makes sense. I was trying to get everything set up before installing it with a 12V supply, but I guess I'll need to do that first, or at least tap the button to wake it.

martinling, 2023-05-25:
I pressed the button when starting up the Pi so that the MCU would be in the right state, and now I2C is reliable. Thanks!

closed 2023-05-25 by martinling