uugear/Witty-Pi-4 #10: User is hardcoded to pi

askpatrickw, 2023-06-01:
https://github.com/uugear/Witty-Pi-4/blob/26e333b996685de9672de45f20e99533bfadd88c/Software/install.sh#L129

The user could be (should be) something other than pi.

nathanael-naeri, 2024-01-11:
As I understand it, this line in the install script:

https://github.com/uugear/Witty-Pi-4/blob/26e333b996685de9672de45f20e99533bfadd88c/Software/install.sh#L129

replaces the hard-coded path in the init script template:

https://github.com/uugear/Witty-Pi-4/blob/26e333b996685de9672de45f20e99533bfadd88c/Software/wittypi/init.sh#L17

by the actual path to the `daemon.sh` file in the `wittypi/` directory, whose path was previously found and stored in the `DIR` variable. And then the result becomes the actual init script `/etc/init.d/wittypi`.

So if the user is not `pi`, or if the install script is run in a different directory than the user's home directory, the result will still be an init script that points to the `daemon.sh` file, wherever it is located (as long as nothing breaks in the logic that populates the `DIR` variable).

uugear, 2024-05-13:
Exactly as [nathanael-naeri](https://github.com/nathanael-naeri) said, this has been considered.

closed 2024-05-13 by uugear