uugear/Witty-Pi-4 #11: raspi-gpio: command not found

jbuck2005, 2023-07-27:
installed software via install.sh

investigating /var/log/syslog I find"
```
Jul 27 19:08:36 pinamehere wittypi[262]: /home/piuser/wittypi/gpio-util.sh: line 231: raspi-gpio: command not found
Jul 27 19:08:36 pinamehere wittypi[262]: /home/piuser/wittypi/daemon.sh: line 155: [: ==: unary operator expected
Jul 27 19:08:37 pinamehere wittypi[262]: /home/piuser/wittypi/gpio-util.sh: line 231: raspi-gpio: command not found
Jul 27 19:08:37 pinamehere wittypi[262]: /home/piuser/wittypi/daemon.sh: line 155: [: ==: unary operator expected
Jul 27 19:08:39 pinamehere wittypi[262]: /home/piuser/wittypi/gpio-util.sh: line 231: raspi-gpio: command not found
Jul 27 19:08:39 pinamehere wittypi[262]: /home/piuser/wittypi/daemon.sh: line 155: [: ==: unary operator expected
```

I've manually installed wiringPi via:
```
wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
```

cat /etc/os-release
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

uugear, 2023-07-29:
I believe raspi-gpio is by default included in Raspbian. In case it is not, you may install it by following the instructions [here](https://github.com/RPi-Distro/raspi-gpio#readme).

The purpose of gpio-util.sh is to provide a substitute of the "gpio" utility in the wiringPi, because the wiringPi is not available since Bullseye. In an older Raspbian version that wiringPi is still available (such as Buster), it will be better to directly use wiringPi instead of gpio-util.sh. To do so you just need to comment out the lines the refer gpio-util.sh, for Witty Pi 4's software they are [here](https://github.com/uugear/Witty-Pi-4/blob/main/Software/wittypi/daemon.sh#L14) and [here](https://github.com/uugear/Witty-Pi-4/blob/main/Software/wittypi/wittyPi.sh#L23).

jbuck2005, 2023-07-29:
While your belief may hold true in most cases, it is false in mine. When I installed the distro for my Pi Zero W, I used the 'lite' version of Raspbian and evidently the raspi-gpio binary is -not- included.

Might I suggest a sanity check routine in your install.sh script that looks for this binary and installs it automatically if missing? This would make your product more accessible for others who may not be comfortable with the command prompt.

uugear, 2023-07-31:
This would be an enhancement, but its priority will be rather low.

jbuck2005, 2023-07-31:
resolved - will look at coding sanity check portion

closed 2023-07-31 by jbuck2005

nathanael-naeri, 2024-01-10:
I am running in the exact same problem as OP on a Raspberry Pi 4B running Ubuntu Server 20.04 LTS.

This is a problem because it makes it impossible to run the Pi on a schedule. Upon writing or choosing a schedule file from the `schedules/` subdirectory, the software and its log files `wittyPi.log` and `schedule.log` report that the next shutdown and startup are appropriately scheduled, but the Pi does not get turned off/on when the time comes, and nothing gets written in the log files to explain why, only in the system logs (e.g. `systemctl status wittypi.service`), so it took me forever to figure out what was going on, as I initially thought my schedule file syntax was bad.

Fortunately `sudo apt install wiringpi` provides the `gpio` command for Ubuntu 20.04 LTS and 22.04 LTS, and I can then disable usage of `gpio-util.sh` by the software as you indicate.

Some heads-up would be nice though - if not in the code in the form of a sanity check and a warning in the logs, at least in the (otherwise excellent) PDF documentation.