uugear/Witty-Pi-4 #16: WittyPi4 Mini does not turn on after LV_SHUTDOWN

jiggak, 2023-10-15:
My goal is to use 5v from USB on my TV connected to the WittyPi4 Mini's `VIN` pin to turn the Pi on when the TV is powered on, and shutdown the Pi when the TV is powered off. I have had this use case working very reliably in the past with the WittyPi3 Mini, with [a few firmware tweaks](https://github.com/jiggak/Witty-Pi-3/tree/vin-usb). But with the addition of `I2C_CONF_IGNORE_POWER_MODE` I thought maybe this would be achievable with the stock WittyPi4 code.

Environment details:
WittyPi4 Mini
Raspberry Pi 4b
Libreelec 11.03
`dtparam=i2c1=on` set in `/boot/config.txt`

I am **not** using the WittyPi scripts, but instead use a simplified script that [just handles shutdown](https://gist.github.com/jiggak/8b2a285a944c70b503bdf4eb743db8aa).

Using the latest firmware revision 5, I have the following registers configured:

```bash
I2C_CONF_LOW_VOLTAGE       = 0x14 # 2v
I2C_CONF_RECOVERY_VOLTAGE  = 0x28 # 4v
I2C_CONF_IGNORE_POWER_MODE = 1
```

With the Pi running, and 5v connected to `VIN`, I can disconnect the 5v source and the Pi will gracefully shutdown. However, once it's shutdown and I attempt to press the power button on the WittyPi, the red power light on the WittyPi and Raspberry Pi briefly turn on and then off again. Pressing the button again and again has the same effect. Disconnecting and reconnecting power doesn't help either.

In this state, I haven't found a way to get the WittyPi to power the Raspberry Pi back on. The only way I've found to recover from this is to power the Raspberry Pi directly to boot it up, set `I2C_CONF_IGNORE_POWER_MODE` back to zero, and then use the power button on the WittyPi to cause the Pi to shutdown gracefully. Then move the power connector from the Pi back to the WittyPi. After this procedure, the WIttyPi can successfully turn the Pi back on with a power button press.

Something strange I noticed is that when I boot the Pi to recover the WittyPi state, using `i2cdump -y 1 0x08` will randomly freeze for a second and fail to read a few bytes. If I repeatedly run `i2cdump` the freeze will occur at different addresses each time. It's seems like the MCU is in some sort of crash/restart loop, but I honestly don't know what's really happening here.

Before attempting to use `LV_SHUTDOWN`, the WittyPi is able to boot and shutdown the Pi as expected with power button press. With the Pi booted in this state, I didn't observe the weird freeze/fail with `i2cdump` as described above. Repeated runs of `i2cdump` are able to output all bytes reliably without any freezing or read failures.

jiggak, 2023-10-15:
I'm not sure if this is relevant or not. I've been trying various changes to the firmware, and after uploading via Arduino IDE, the WittyPi is unable to turn on the Pi from a button press. The same thing as described above occurs: the red power LED turns on briefly then off again. I have to power the Pi on directly to boot the Pi. Again, `ic2dump` has the same random freeze/read errors described above but I can see that the registers have all been reset to defaults.

Once I shutdown the Pi using the WittyPi button, connect the power back to the WittyPi, the WittyPi is then able to boot and shutdown the Pi as expected. This happens even after I revert any changes I made to the firmware, and upload from Arduino IDE.

uugear, 2023-10-17:
When you are using modified firmware and modified software, it will be very difficult for us to give you advice. Even if you provide all information of what you have done, we do not have the bandwidth to reproduce this issue on our end.

The strange thing you noticed when using i2cdump -y 1 0x08, most probably was due to the fact that MCU (ATtiny841) was in sleep mode (white LED blinking). If you tap the button just after powring your Pi, this problem should disappear. However if you tap the button after your Pi finish the boot (daemon.sh is running), the software will shutdown your Pi.

jiggak, 2023-10-17:
The observations I made are with unmodified firmware. I only tried firmware changes in an attempt to diagnose the issue, but I quickly reverted back to the stock firmware.

Are you able to confirm my attempted use case would be possible with stock firmware? Using `VIN` pin to detect when 5v is present and power on the Pi, and then detect when 5v is disconnected and initiate shutdown?

Lastly, is there any tips you can suggest for debugging the firmware? Is it possible to get a serial monitor connected somehow (e.g. with 3.3v FTDI adapter perhaps)?

uugear, 2023-10-20:
I just made a test on Witty Pi 4 Mini with official firmware, and it works perfectly with your use case.

I use our software. I only set one I2C register:
`i2cset -y 1 8 41 1`

Then I configure the low voltage threshold to 3V and recovery voltage to 4V. I connect 5V to VIN on Witty Pi 4 Mini.

If I turn off the 5V on VIN, Witty Pi 4 Mini turns off the Pi. If I turn on the 5V on VIN, Witty Pi 4 Mini turns on the Pi.

In our lab, we debug the firmware by outputing debugging information to the internal I2C bus, and we made an I2C debugger and hook it to the unpopulated P5 on Witty Pi 4 Mini. That I2C debugger read data from I2C bus and forward it to the serial port, so we can read them on PC via a serial port adapter.

jiggak, 2023-10-21:
> In our lab, we debug the firmware by outputing debugg information to the internal I2C bus, and we made an I2C debugger and hook it to the unpopulated P5 on Witty Pi 4 Mini. That I2C debugger read data from I2C bus and forward it to the serial port, so we can read them on PC via a serial port adapter.

Thanks for this! This gives me something to research.

Your results sounded promising so I just had to try again. In my original tests I was using LibreELEC with my own shutdown script. I can't imagine that would be causing the MCU to bug out the way it is, but I thought I'd try again and take that out of the equation.
1. Flash latest WittyPi4 Mini firmware https://github.com/uugear/Witty-Pi-4/blob/main/Firmware/WittyPi4/WittyPi4_Mini.ino.hex
2. Connect 5V and GND from USB on my PC to VIN and GND on WittyPi Mini
3. Connect USB-C (3A) to WittyPi Mini, boot Pi using button on WittyPi
4. Used latest "Raspberry Pi OS (64-bit)"
5. Installed WittyPi software using the `install.sh` file via wget
6. After flashing all WittyPi settings appear to be reset (EPROM got erased during flash I guess?)
7. I used `wittyPi.sh` to set low voltage threshold to 3V and recovery voltage to 4V
8. I used `ic2set -y 1 8 41 1`
9. Run `daemon.sh` as root
10. Unplug USB (cut 5V from VIN)
11. Pi shuts down gracefully as expected, red LED's turn off and white LED begins to blink on WittyPi.
12. Reconnect USB. The white LED flashes, then red LED's flash on both WittyPi and Pi, Pi doesn't turn on. This repeats every 4 secs it appears.
13. Press button on WittyPi and the red LED's both flash but the Pi doesn't turn on.

The only way I found so far to get out of this state is to connect USB-C to boot Pi. Run the `daemon.sh` script, and press the button on the WittyPi to cause the Pi to shutdown. After this I can put the USB-C back to the WittyPi and use the power button to boot the Pi again.

I used the following command to flash the firmware using a USBtinyISP.
`~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -c usbtiny -p attiny841 -C ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -Uflash:w:./WittyPi4_Mini.ino.hex:i`

jiggak, 2023-10-21:
Oh, another thing. After I boot the Pi directly with USB-C, and I look at the value of `I2C_ACTION_REASON`, it's value is 5 (Input voltage is restored). Not sure if that's relevant but it's the only thing that stood out from `i2cdump`.

jiggak, 2023-10-23:
I have an interesting development. I stumbled upon this post in your forums.

https://www.uugear.com/forums/technial-support-discussion/witty-pi-4-tapping-the-on-off-switch-doesnt-turn-on-the-raspberry-pi/

I have one of the older WittyPi4 Mini's without the filter cap so I tried adding one (I used a 47uF electrolytic) and it seems to have fixed the issue I experienced. The wittypi will now reliably turn the Pi on after a previous LV shutdown.

closed 2023-11-01 by jiggak