uugear/Witty-Pi-4 #18: how do you change the default listen port using the TXD on gpio14

newpond, 2023-11-03:
I have a zwave hat that uses this port and when I shutdown the pi, its boots up again as the hat is obviously keeping the gpio high.
Can this port be moved to something else?

uugear, 2023-11-06:
Unfortunately this pin can not be moved.

closed 2023-11-06 by uugear

MarcDerhammer, 2023-11-08:
> I actually managed to resolve this by adding the following line to the /boot/config.txt (to set the pin low on shutdown)
> 
> dtoverlay=gpio-poweroff,gpiopin=14,active_low="y"
> 
> And then adding the following line to /home/pi/witty/daemon.sh (to set it back to normal at boot up) gpio -1 mode 8 alt0

Thank you, this worked for me as well

newpond, 2023-11-10:
> Thank you, this worked for me as well

you're welcome.
out of interest could try timing the shutdown of your pi (until the green goes off) with these commands installed and without them installed.
It does appear on my system (running Domoticz) that it shuts down quicker by 4-5  seconds with these command added, which makes me wonder whether it is doing a full shutdown correctly, although I can't see anything in the log that would imply this is the case.
Also what setting do you use for "cut power delay" as I have read that the kernel can panic using the power down command if the power is not cut within 3 seconds, again I cannot see anything in the logs that would imply this is happening even when I set this to 5 seconds or more.
cheers