uugear/Witty-Pi-4 #21: Raspberry Pi OS Bookworm bug

michaelsage, 2023-12-13:
Hi,
I've been running the wittypi 4 on bookworm successfully for a number of weeks. However, an update appears to have broken the wittypi scheduler. I suspect something has changed with the way the Pi is using the GPIO. The scheduled startup and shutdown have stopped updating. 

I think I have managed to track down the problem to the following within the wittypi service (repeated every second):
Dec 13 10:03:51 studykioskpi wittypi[589]: /home/pi/wittypi/daemon.sh: line 155: [: too many arguments

In the file this is related to: # delay until GPIO pin state gets stable

I am also experiencing this problem with a wittypi 3 mini on a Pi Zero 2WH (it's the same section in the file but a different line number (119)).

I am unsure if this is the right issue as I don't have the logs before the latest round of updates, but suspect it probably is?

michaelsage, 2023-12-15:
systemctl status wittypi.service shows the error every second it will give you a line number for your daemon.sh file. If you open the file using nano (or similar) then comment out that whole section (it's called delay until pin state). Then it works, but I'm not 100% sure what that section does...

uugear, 2024-05-13:
Has been fixed in V4.14

closed 2024-05-13 by uugear