uugear/Witty-Pi-4 #23: What is the "Default ON delay" setting and how can I clear settings?

nathanael-naeri, 2024-01-11:
The setting 9 "Default ON delay" in the `wittyPi.sh` script, screenshot below (sorry for blurry photo), is not documented in the PDF documentation under section 7.11 "View/change other settings". What does this setting mean?

[image]

Also, are the settings stored in a file I can delete to easily revert to the defaults? Choosing option 12 "Reset data" in the `wittyPi.sh` script clears scheduled times and thresholds, but does not reset the settings, as far as I can tell.

3ricj, 2024-01-11:
If you enable setting 1 (Default state when powered), then the setting 9 causes a delay when powered on.   So it will wait for N seconds after getting power before turning on the system.

uugear, 2024-03-18:
This setting has been documented on page 14 in the [user manual](https://www.uugear.com/doc/WittyPi4_UserManual.pdf).

Questions about how to use the product should be asked in [the support forum](https://www.uugear.com/forums/technial-support-discussion/) instead.

closed 2024-03-18 by uugear