uugear/Witty-Pi-4 #39: Hard reboot

c17932, 2024-12-05:
Is there any way to hard reboot the pi (with witty pi 4 l3v7) if the pi hangs and the witty pi doesn’t shut down? Your device would work perfectly, even remotely when hooked up to a smart plug, if there was.

I was thinking you could do a simple, power-dependent thing like power cycle twice (while getting nothing over gpio’s from the pi) to hard reboot…

uugear, 2024-12-06:
This can be done by modifying the firmware, to implement a watchdog functionality. The software can keep sending heartbeat signal to Witty Pi via GPIO or via I2C (update a specific register), the firmware tracks the heartbeat signal and does a power cycle following an emulated button tap, which will make sure to turn on the Pi again.

It is not that difficult, the problem is the current firmware has used up almost all programing space in ATtiny841. You will have to delete some functionalities to reserve the space for this new function.

c17932, 2024-12-06:
So I looked into it, and yes, the witty pi has a mechanism to watch the pi's txd pin for activity and shut down if there is none, so it's *perfect* (works remotely)...case closed! :)

closed 2024-12-06 by c17932